tpaviot/pythonocc-core #1341: Function BRepTools_History.Merge() not wrapped

ChristianHesse, 2024-06-03:
I tried to use the function BRepTools_History.Merge() and get the following error:
MethodNotWrappedError: Merge not wrapped

I tried it using Versions 7.7.2 and 7.8.1 with the same result. Are there any plans to support it?

tpaviot, 2024-06-03:
You're right, it is excluded from the wrapper, but I don't remember why. Could you please post a small code snippet I could use to test the wrapper?

ChristianHesse, 2024-06-03:
Thanks for the quick reply. Sure, here is a small snippet:
```
from OCC.Core.gp import gp_Pnt, gp_Ax2, gp_Dir
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox, BRepPrimAPI_MakeCylinder
from OCC.Core.BRepAlgoAPI import BRepAlgoAPI_Fuse
from OCC.Core.BRepTools import BRepTools_History

# Create the first shape: a box
box = BRepPrimAPI_MakeBox(100, 100, 100).Shape()

# Create the second shape: a cylinder
cylinder = BRepPrimAPI_MakeCylinder(gp_Ax2(gp_Pnt(50, 50, 0), gp_Dir(0, 0, 1)), 50, 150).Shape()

# Perform the merge operation
fuse = BRepAlgoAPI_Fuse(box, cylinder)
fuse.Build()

# Get the resulting shape
merged_shape = fuse.Shape()

# Create a history object to track the operation
history = BRepTools_History()
history.Merge(fuse)

# Print out the history of the merge operation
print("History of the merge operation:")
for i in range(history.Generated(box).Length()):
    print(f"Generated from box: {history.Generated(box).Value(i + 1)}")
for i in range(history.Generated(cylinder).Length()):
    print(f"Generated from cylinder: {history.Generated(cylinder).Value(i + 1)}")
```

tpaviot, 2024-06-03:
ok, I created the related wrapper on my machine. The code provided doesn't pass. Are you sure of the use of `BRepTools_History`? It seems the `Merge` method expects another history (this method merges two different histories).

ChristianHesse, 2024-06-03:
You are right. I edited the code to in order to pass another history object to the `Merge` method. The last part I could not test properly yet.

tpaviot, 2024-06-04:
The `Merge` method has been added to the wrapper, I close the issue, please post an example here as soon as you have something working

closed 2024-06-04 by tpaviot

ChristianHesse, 2024-11-14:
Thanks a lot for adding the function to the wrapper. Here is the updated, working code example:

```python
from OCC.Core.gp import gp_Pnt, gp_Ax2, gp_Dir
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox, BRepPrimAPI_MakeCylinder
from OCC.Core.BRepAlgoAPI import BRepAlgoAPI_Fuse
from OCC.Core.BRepTools import BRepTools_History
from OCC.Core.TopAbs import TopAbs_FACE
from OCC.Core.TopExp import TopExp_Explorer
from OCC.Core.TopTools import TopTools_ListIteratorOfListOfShape

# Create the first shape: a box
box = BRepPrimAPI_MakeBox(100, 100, 100).Shape()

# Create the second shape: a cylinder
cylinder = BRepPrimAPI_MakeCylinder(gp_Ax2(gp_Pnt(50, 50, 0), gp_Dir(0, 0, 1)), 50, 150).Shape()

# Perform the merge operation
fuse = BRepAlgoAPI_Fuse(box, cylinder)
fuse.Build()

# Get the resulting shape
merged_shape = fuse.Shape()

# Create a history object to track the operation
history = BRepTools_History()
history.Merge(fuse.History())

def iter_type(shp, type=TopAbs_FACE):
    exp = TopExp_Explorer(shp, type)
    while exp.More():
        res = exp.Current()
        yield res
        exp.Next()

def iter_list_of_shape(list_of_shape):
    occ_iterator = TopTools_ListIteratorOfListOfShape(list_of_shape)
    while occ_iterator.More():
        yield occ_iterator.Value()
        occ_iterator.Next()

# Print out the history of the merge operation
print("History of the merge operation:")
for fi, face in enumerate(iter_type(box)):
    for i in range(history.Generated(face).Size()):
        print(f"Generated from box face no {fi}: {[shape for shape in iter_list_of_shape(history.Generated(face))]}")
for fi, face in enumerate(iter_type(cylinder)):
    for i in range(history.Generated(face).Size()):
        print(f"Generated from cylinder face no {fi}: {[shape for shape in iter_list_of_shape(history.Generated(face))]}")
```