tpaviot/pythonocc-core #1344: thher is something wrong when use version7.8.1 with pyside6

whitecateattea, 2024-06-27:
[image]
when I run the program with pysdie6 and pythonocc 7.8.1, the screen have the message box above, but pyqt5 or pythonocc old version is ok.

whitecateattea, 2024-06-27:
> I use Pyside6 and pythonocc7.8.1 with no problems, what is your code?

oh, I have solved it, the reason is my anaconda, so I changed a new one and it's ok. thanks anyway.

djy89, 2024-07-21:
> I use Pyside6 and pythonocc7.8.1 with no problems, what is your code?
Can you tell me which anaconda verison and pycharm version you use?

wzhsylm, 2024-08-01:
How did you solve it? I also encountered the same error