tpaviot/pythonocc-core #1345: was HashCode removed from TopAbs_Face?

philosophical1337, 2024-06-27:
I am working with step files and when imported using read_step_file_with_names_colors you can always tell they contain many duplicate faces (obvious z-fighting). 

I'm not sure if its just not noticeable when importing without colors due to the surfaces all being grey but anyway, I was using the following function to compare all the faces and discard duplicates which was working wonderfully until I updated to 7.8.1. 

now I get the error message that HashCode attribute doesn't exist.

```
identifier = face.HashCode(-1)
                   ^^^^^^^^^^^^^
AttributeError: 'TopoDS_Face' object has no attribute 'HashCode'
```

here is my function:

```
    def removeDuplicateFaces(self, shape):
        # Create a compound to hold the result
        compound = TopoDS_Compound()
        builder = BRep_Builder()
        builder.MakeCompound(compound)

        uniqueFaces = {}

        explorer = TopExp_Explorer(shape, TopAbs_FACE)
        while explorer.More():
            face = explorer.Current()

            identifier = face.HashCode(-1)

            if identifier not in uniqueFaces:
                # Add the face to the result compound
                builder.Add(compound, face)
                # Store the identifier
                uniqueFaces[identifier] = face

            explorer.Next()

        return compound
```

tpaviot, 2024-06-28:
The HashCode method was actually removed from OpenCascade itself. You can use the builtin `hash` function:
 ```identifier=hash(face)```

tpaviot, 2024-06-28:
I should have generated a deprecation warning when using the HashCode method

philosophical1337, 2024-06-28:
this is just a guess but although the faces are geometrically identical, i think the actual face objects have small differences, so just hashing the object with the built in hash doesn't work at all.  

I think the HashCode method was hashing only the geometric properties of the face and thus worked to find the identical faces, ignoring the differences in the object

philosophical1337, 2024-06-28:
okay, i didn't realize that my code was effectively always returning only the first face and then discarding the rest, the HashCode(-1) just resulted in a hash of 1 every time.

i just ended up discarding everything but the first face and it works out the same, oops!

closed 2024-11-16 by tpaviot