tpaviot/pythonocc-core #1350: String not interfaced to IStream

JJMinton, 2024-07-10:
Updating from 7.5.1 to 7.8.1, I am trying to replace `BRepTools_ShapeSet.ReadFromString` with `BRepTools_ShapeSet.Read` as I believe is the result of #1301 .

The resulting error was:
```python
>>> from OCC.Core.BRepTools import BRepTools_ShapeSet
>>> BRepTools_ShapeSet().Read("foo")
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/miniconda3/envs/occwl_dev/lib/python3.9/site-packages/OCC/Core/TopTools.py", line 2880, in Read
    return _TopTools.TopTools_ShapeSet_Read(self, *args)
TypeError: Wrong number or type of arguments for overloaded function 'TopTools_ShapeSet_Read'.
  Possible C/C++ prototypes are:
    TopTools_ShapeSet::Read(std::istream &,Message_ProgressRange const &)
    TopTools_ShapeSet::Read(TopoDS_Shape &,std::istream &)
```

I am installing from conda in a docker container:
```Dockerfile
# syntax=docker/dockerfile:1.4
FROM --platform=linux/amd64 python:3.9 as base

RUN python -m pip install --upgrade pip

FROM base as dev

# Insgtall OCC requirements
RUN apt-get update
RUN  apt-get install -y ffmpeg libsm6 libxext6

# Install conda
RUN mkdir -p ~/miniconda3 && \
    wget https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O ~/miniconda3/miniconda.sh && \
    bash ~/miniconda3/miniconda.sh -b -u -p /miniconda3 && \
    rm -rf ~/miniconda3/miniconda.sh && \
    /miniconda3/bin/conda init bash

ARG PROJ_DIR=/proj
ARG PIP_CACHE=/root/.cache/pip
WORKDIR ${PROJ_DIR}

# Create conda environment
COPY environment.yml .
RUN --mount=type=cache,target=/miniconda3/pkgs,id=conda \
    /miniconda3/bin/conda env create -f environment.yml
```
where `environment.yml` is
```yaml
name: occwl_dev
channels:
  - conda-forge
  - defaults
  - anaconda
dependencies:
  - python=3.9
  - numpy
  - pythonocc-core=7.8.1
  - networkx
  - wxPython
  - pydeprecate
```

The full context is [this PR](https://github.com/AutodeskAILab/occwl/pull/36/files)

tpaviot, 2024-07-12:
Confirmed. Under investiagtion. If your need is to load a TopoDS_Shape from a serialization, you can try:

```python
from OCC.Core.BRepTools import breptools
the_shape = breptools.ReadFromString("foo")
```
this is how shape pickling is implemented for the TopoDS_Shape class.

Andrej730, 2024-07-15:
@tpaviot 

1) Can you please help narrow down, from which OCC or PythonOCC version this change occurred?

2) Is new way

```python
from OCC.Core.BRepTools import breptools
brep_string = ""
shape = breptools.ReadFromString(brep_string)
```

is equivalent to the old snippet below or there are some caveats? 

This snippet seems to be used often by users - https://github.com/search?q=BRepTools_ShapeSet+ReadFromString+NbShapes+language%3APython&type=code

```python
from OCC.Core.BRepTools import BRepTools_ShapeSet
brep_string = ""
shape_set = BRepTools_ShapeSet()
shape_set.ReadFromString(brep_string)
shape = shape_set.Shape(shape_set.NbShapes())
```

JJMinton, 2024-07-23:
Thanks for the response @tpaviot, any progress with this? Is there any help I could provide?
Unfortunately, we're looking for all the subshapes so can't make do with `breptools`.

tpaviot, 2024-07-23:
@JJMinton I can fix the regression by just resetting the `ReadFromString` method as it used to be before the latest release, it will be published in the next release (the 7.8.2 release of occt might come in a few days/weeks).

Note that the `ReadFromString` method did use `breptools`:

```c++
static TopoDS_Shape ReadFromString(const std::string & src) {
   std::stringstream s(src);
   TopoDS_Shape shape;
   BRep_Builder b;
   BRepTools::Read(shape, s, b);
   return shape;
}
```

commit 02f8ddf2 (2024-11-14): Add back BRepTools_ShapeSet ReadFromString and WriteToString methods to fix regressions. Fix #1350
closed 2024-11-14 by tpaviot