tpaviot/pythonocc-core #1357: import numpy  >=1.26.4, raise error

djy89, 2024-07-21:
when i add import numpy as np at file.py  and run debug
why raise this.....
File "....python3.12/site-packages/OCC/Display/qtDisplay.py", line 104, in InitDriver
    self._display.Create(window_handle=int(self.winId()), parent=self)
  File ".../python3.12/site-packages/OCC/Display/OCCViewer.py", line 212, in Create
    self.Init(self._window_handle)
  File ".../python3.12/site-packages/OCC/Core/Visualization.py", line 185, in Init
    return _Visualization.Display3d_Init(self, handle)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Aspect_GraphicDeviceDefinitionErrorOpenGl_Window::CreateWindow: glXCreateContext failed. raised from method Init of class Display3d

tpaviot, 2024-07-22:
please attach the related file

djy89, 2024-07-22:
Thank you for your attention. For all the imported packages in this file, no matter how they are coordinated, importing numpy will report errors related to OpenGL. In addition, when I use load_mackend ("pyqt6"), these errors will also be reported. How can I determine if I will only use pyqt6 and importing numpy without errors, or if I will only use pyside6 and importing numpy without errors

djy89, 2024-07-22:
from __future__ import print_function
import sys
from OCC.Display.backend import load_any_qt_backend, get_qt_modules
load_any_qt_backend()
from OCC.Display.backend import load_backend
load_backend("pyside6")
from PyQt6 import QtWidgets, QtGui, QtCore, QtWidgets, QtOpenGL
# from PySide6 import QtWidgets, QtGui, QtCore, QtWidgets, QtOpenGL
# QtCore, QtGui, QtWidgets, QtOpenGL = get_qt_modules()
from OCC.Display.qtDisplay import qtViewer3d
# import numpy  as np

class GLWidget(qtViewer3d):
    def __init__(self, parent=None):
        super(GLWidget, self).__init__(parent)
if __name__ == '__main__':
    def TestOverPainting():
        class AppFrame(QtWidgets.QWidget):
            def __init__(self, parent=None):
                QtWidgets.QWidget.__init__(self, parent)
                self.setWindowTitle(self.tr("qtDisplay3d overpainting example"))
                self.resize(1280, 1024)
                self.canva = GLWidget(self)
                mainLayout = QtWidgets.QHBoxLayout()
                mainLayout.addWidget(self.canva)
                mainLayout.setContentsMargins(0, 0, 0, 0)
                self.setLayout(mainLayout)

            def runTests(self):
                self.canva._display.Test()

        app = QtWidgets.QApplication(sys.argv)
        frame = AppFrame()
        frame.show()
        frame.runTests()
        app.exec()

    TestOverPainting()

djy89, 2024-07-22:
python3.11  and pythonocc-core 7.8.1 use only pyside6  and use only qyqt6  raise about opengl display errors  when import numpy