tpaviot/pythonocc-core #1358: TraitError: The 'rotation' trait of a GridHelper instance contains an Enum of an Euler which expected any of ['XYZ', 'YZX', 'ZXY', 'XZY', 'YXZ', 'ZYX'], not the str 'xyz'.

SGR-FA, 2024-07-29:
Hi,

First of all, thank you all for your great work !!

I'm upgrading from an old Python 3.8 venv but unfortunately I get this message when I try to open a viewer in my notebooks.
For now, I'm completely stucked but as I understood, the "XYZ" value of the Euler rotation gets lowercased somewhere during the Jupyter traits/session/zmqstream/socket communication. It could be during any of the cast, from_json, to_json, send_multipart or recv_multipart steps but I have not managed to find which one :( and I get the same TraitError for any other version of python/occt/ifcopenshell...

Does anyone also experience this or did I miss something?

Here is the venv :

```
#conda env export --from-history

name: ifc_viewer_311
channels:
  - defaults
dependencies:
  - pythonocc-core[build=*novtk*]
  - lark
  - occt=7.7.2
  - ifcopenshell
  - python=3.11
  - jsonpath-ng
  - pythreejs
  - vpython
  - cmasher
  - ipywidgets
  - ipympl
  - matplotlib
  - traitlets
  - jupyterlab
  - scipy
  - pandas
  - tornado
```
Here is an example to get the error:
```
import ifcopenshell
import ifcopenshell.geom

settings = ifcopenshell.geom.settings()
settings.set(settings.USE_PYTHON_OPENCASCADE, True)

from OCC.Display.WebGl.jupyter_renderer import JupyterRenderer as Renderer

class ifc_viewer(Renderer):
    def __init__(self):
        Renderer.__init__(self)
        self.n = 0
        self.products = []

    @staticmethod
    def subshapes(shp):
        import OCC.Core.TopoDS
        it = OCC.Core.TopoDS.TopoDS_Iterator(shp)
        while it.More():
            yield it.Value()
            it.Next()

    def DisplayShape(self, product, shp, color):
        for shp, sty in zip(self.subshapes(shp), color):
            Renderer.DisplayShape(self, shp, shape_color=sty[0:3], render_edges=True, transparency=True)
            self.products.append(product)

    def onclick(self, item):
        self.html.value = "Selected %r" % self.products[int(item.name)]

file = ifcopenshell.open("YOUR_IFC_FILE.ifc")
geometry = dict((file[item.data.id], (item.geometry, item.styles)) for item in ifcopenshell.geom.iterator(settings, file))

viewer = ifc_viewer()

for product, (shape, styles) in geometry.items():
    viewer.DisplayShape(product, shape, styles)
    
viewer.Display()
```

Thank you for your help.

wll0307, 2024-09-23:
TraitError: The 'rotation' trait of a GridHelper instance contains an Enum of an Euler which expected any of ['XYZ', 'YZX', 'ZXY', 'XZY', 'YXZ', 'ZYX'], not the str 'xyz'.
我也是这个报错你是用啥解决的

SGR-FA, 2024-09-23:
Hi,
(please translate, I greatly lack ability to speak/write in this language)
Unfortunately, trials and errors forced me to use the ```rotateX``` method.
```
class CustomJupyterRenderer(JupyterRenderer):
...
        self.my_grid = Grid(bb_center=self._bb.center, maximum=bb_max,
                                    colorCenterLine='#aaa', colorGrid='#ddd')
...
        # raise error for now: self.my_grid.set_rotation(((math.pi / 2.0), 0, 0, "XYZ"))
        self.my_grid.grid.rotateX(math.pi / 2.0)
...
```
BR

SGR-FA, 2024-10-29:
> TraitError: The 'rotation' trait of a GridHelper instance contains an Enum of an Euler which expected any of ['XYZ', 'YZX', 'ZXY', 'XZY', 'YXZ', 'ZYX'], not the str 'xyz'. 我也是这个报错你是用啥解决的

FYI, there is an ongoing fix thanks to @vidartf : see [issue on jupyter-widgets/pythreejs](https://github.com/jupyter-widgets/pythreejs/issues/413)

wll0307, 2024-12-04:
> > TraitError：GridHelper实例的“旋转”特征包含Euler的枚举，需要它['XYZ'，'YZX'，'ZXY'，'XZY'，'YXZ'，'ZYX']中的任何一个，而不是str 'xyz'。我也是这个报错你是用啥解决的
> 
> 仅供参考，目前发售修复版，感谢@vidartf：请参阅[jupyter-widgets/py Threejs 上的问题](https://github.com/jupyter-widgets/pythreejs/issues/413)
你好，现在这个bug修复了吗？