tpaviot/pythonocc-core #1363: Please destroy the QApplication singleton before creating a new QApplication instance.

LZHLZHOOO, 2024-08-18:
[image]
I want to upload a 3D model in STP format, but I'm encountering this error. Does anyone know what might be causing this issue? Please help.

jnwalther, 2024-10-31:
I'm not 100% sure if this is the same issue, but I got the same error message when trying to open a display window for a second in a single Python session. It seems to be related to Pyside6. Other projects seem to have [similar issues](https://github.com/runesc/PPG/issues/7).

The solution given in that issue seems to do the trick for me, i.e. go to https://github.com/tpaviot/pythonocc-core/blob/master/src/Display/SimpleGui.py and do
```diff
-  app = QtWidgets.QApplication(sys.argv)
+  app = QtWidgets.QApplication.instance() or QtWidgets.QApplication(sys.argv)
```
in line 230.

I haven't tested it completely, so I can't say if this change breaks anything for the other Qt versions.

tpaviot, 2024-11-15:
Fixed by #1381

closed 2024-11-15 by tpaviot