tpaviot/pythonocc-core #1367: Disable log printing when calling write_step_file

GiovaGa, 2024-09-04:
Is there any way to disable the following log:
```
*******************************************************************
******        Statistics on Transfer (Write)                 ******

*******************************************************************
******        Transfer Mode = 0  I.E.  As Is       ******
******        Transferring Shape, ShapeType = 0                      ******
** WorkSession : Sending all data
 Step File Name : [...] (31535 ents)  Write  Done
```

It is printed every time I write a file. As I am doing some batch jobs, it could hide some more useful messages.

If it matters I am working on Windows 11 (64 bit).

Tanneguydv, 2024-09-04:
Here a simple code to bypass the logs:

```python
import os
import sys
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from  OCC.Extend.DataExchange import write_step_file

def create_box():
    # Create a simple box with dimensions 10x20x30
    return BRepPrimAPI_MakeBox(10.0, 20.0, 30.0).Shape()

def execute_without_print(func, *args, **kwargs):
    # Save original stdout and stderr
    stdout_fd = sys.stdout.fileno()
    stderr_fd = sys.stderr.fileno()

    # Create a temporary file to capture the output
    with open(os.devnull, 'w') as devnull:
        # Redirect stdout and stderr to /dev/null
        os.dup2(devnull.fileno(), stdout_fd)
        os.dup2(devnull.fileno(), stderr_fd)

        try:
            # Execute the function
            return func(*args, **kwargs)
        finally:
            # Restore stdout and stderr
            sys.stdout = sys.__stdout__
            sys.stderr = sys.__stderr__

# Example 
box = create_box()

# Execute write_step_file without printing any messages
execute_without_print(write_step_file, box, "box.step")

print("STEP file has been written successfully.")
```

GiovaGa, 2024-09-05:
Not really what I was expecting, but it surely does what I need.
I still think it could be useful to have a parameter in write_step_file (and probably other functions too) to disable logging.

Does depend on the OpenCascade C++ code?
Otherwise I can try to contribute to this goal

Tanneguydv, 2024-09-05:
Indeed, it depends on the C++ code

GiovaGa, 2024-09-05:
Ok, then I guess this issue can be closed. Thank you for the support

closed 2024-09-05 by GiovaGa

raovgarimella, 2025-04-27:
This works very well when turning off the output but doesn't seem to turn it back on

gkv311, 2025-06-03:
These messages come through [`Message_Messenger` interface](https://stackoverflow.com/questions/64912439/preventing-opencascade-to-write-to-console/64915870#64915870) in OCCT.

If you are not interested in informational messages, you may tune gravity of messages to be printed through global messenger.
Gravity levels defined by enumeration `Message_Gravity`:
- `5` - display nothing;
- `Message_Fail` - display only failures;
- `Message_Alarm` - display only failures and non-breaking errors;
- `Message_Warning` - display failures and warnings;
- `Message_Info` - display failures, warnings and informational message (like operation success); this is **DEFAULT** level;
- `Message_Trace` - display failures, warnings, informational and verbose messages from algorithms.

Default gravity level is set to `Message_Info`, which means that all messages will be displayed, except verbose ones.
The message from STEP writer `Statistics on Transfer (Write)` that you see has `Message_Info` gravity, which means that it will be suppressed by increasing gravity level to `Message_Warning` (in case if you still want to see warnings) or to ``Message_Alarm` (in case if want to see only errors):

```.py
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.Message import message, Message_Gravity
from OCC.Extend.DataExchange import write_step_file

# configure global OCCT messenger
# Message_Warning
msgGravity = Message_Gravity.Message_Warning
messenger = message.DefaultMessenger()
msgPrinters = messenger.Printers()
for i in range(msgPrinters.Length()):
    printer = msgPrinters.Value(i + 1)
    printer.SetTraceLevel(msgGravity)

# write sample model into STEP
box = BRepPrimAPI_MakeBox(10.0, 20.0, 30.0).Shape()
write_step_file(box, "box.step")
```