tpaviot/pythonocc-core #1371: pyqt5 backend could not be loaded

purushothamanmunusamy, 2024-10-06:
pyqt5 backend could not be loaded.

I used "pip install PyQt5" and the requirement already satisfied in visual studio developer command prompt.

Please help.

Tanneguydv, 2024-10-07:
Have you source your OCC env in your terminal before `pip install pyqt5` ?

purushothamanmunusamy, 2024-10-07:
i don't know how to do it. Can you please guide?

Tanneguydv, 2024-10-08:
Look at [https://pypi.org/project/occ/]()

```
"Octavia Chicken Checker looks for abandoned Octavia load balancer artifacts, amphoras, and so forth. Optionally it will clean them up as well."
````

It doesn't really looks like PythonOCC...

Please refer to the install doc:
[https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md]()