tpaviot/pythonocc-core #1372: ImportError: cannot import name 'Geom_Plane' from 'OCC.Core.GeomAPI'

Bill-XU, 2024-10-09:
Hello,

I'm using PythonOCC on Windows.
I have installed it by conda-forge, and am using Python3.11
Everything is fine, import completion works correctly as well that it prompts there is a class named "Geom_Plane" in OCC.Core.GeomAPI.

But when I ran the code, 
```python
from OCC.Core.GeomAPI import Geom_Plane
```
python gave me this error:
```sh
ImportError: cannot import name 'Geom_Plane' from 'OCC.Core.GeomAPI' (...\.conda\envs\python311\Lib\site-packages\OCC\Core\GeomAPI.py)
```

I have checked "GeomAPI.py", there is no such "Geom_Plane" in it.
Is this a compilation issue? How to fix it?

Best regards,
Bill

Tanneguydv, 2024-10-11:
Try : 

`from OCC.Core.Geom import Geom_Plane
`

Bill-XU, 2024-10-12:
> Try :
> 
> `from OCC.Core.Geom import Geom_Plane `

@Tanneguydv 

Thank you!
(I also see there are related examples in pythonocc-demos, next time will check them first.)

Bill

closed 2024-10-12 by Bill-XU