tpaviot/pythonocc-core #1378: Install pythonocc-core failed in PyCharm

jerrycwong, 2024-10-28:
Hello there,

I am a new to python and PyCharm. I tried to install the package of pythonocc-core in PyCharm and I got the following error codes.

ERROR: Could not find a version that satisfies the requirement pythonocc-core (from versions: none)
ERROR: No matching distribution found for pythonocc-core

My Python version is Python 3.13.0.

I could not find the related issue in the thread, so I hope anyone can help me to resolve the issue.

Appreciate in advance!

Tanneguydv, 2024-10-31:
I suppose you tried `pip install pythonocc-core`, command that does lead nowhere.
To install pythonocc-core simply you just have to follow the https://github.com/tpaviot/pythonocc-core?tab=readme-ov-file#install-with-conda section.

When discovering a new library it's a common use to read the `Readme`, and when there's an issue it's also necessary, for the sake of the community, to refer to the existing issues, see
https://github.com/tpaviot/pythonocc-core/issues?q=is%3Aissue+ERROR%3A+No+matching+distribution+found+for+pythonocc-core+is%3Aclosed

tpaviot, 2024-10-31:
FYI conda-forge packages are only available up to python 3.12

You can downgrade your python version to 3.12 or lower and retry