tpaviot/pythonocc-core #1388: How to colorize a model with gradient colors?

Bill-XU, 2024-11-16:
Hi,

Asked the same question in OCCT's discussions, but got no answer.
Wonder if there is anybody know this here.

I found in OCCT's manual https://dev.opencascade.org/doc/overview/html/index.html#intro_overview_modalgo,
there is a picture of a hexagon nut, which is colorized with linearly gradient colors from red to blue.
[image]

How could I do the same colorization using Python OCC?
Thanks for any helpful idea and hints~~

Best regards,
Bill

tpaviot, 2024-11-16:
Try using GLSL shaders, see https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_visualization_glsl.py

tpaviot, 2024-11-16:
```python
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

from OCC.Core.AIS import AIS_Shape
from OCC.Core.Graphic3d import (
    Graphic3d_ShaderProgram,
    Graphic3d_TOS_VERTEX,
    Graphic3d_TOS_FRAGMENT,
    Graphic3d_ShaderObject,
)
from OCC.Display.SimpleGui import init_display

# Create display
display, start_display, add_menu, add_function_to_menu = init_display()

# Create box
box = BRepPrimAPI_MakeBox(10, 20, 30).Shape()

# Create AIS_Shape
ais_shape = AIS_Shape(box)
display.Context.Display(ais_shape, True)

# Define vertex shader
vertex_shader = """
varying vec3 vPosition;
varying vec3 vNormal;

void main() {
    vPosition = occVertex.xyz;  // Get vertex position
    vNormal = occNormal;        // Get vertex normal
    
    // Standard vertex position calculation
    gl_Position = occProjectionMatrix * occWorldViewMatrix * occModelWorldMatrix * occVertex;
}
"""

# Define fragment shader
fragment_shader = """
    varying vec3 vPosition;
    varying vec3 vNormal;
    
    void main() {
        // Adjust the normalization range to favor red more
        float normalizedZ = (vPosition.z + 20.0) / 40.0;
        
        // Apply a power function to bias towards red
        float t = pow(normalizedZ, 1.5);  // Increased power for more red
        
        // Create gradient with enhanced red
        vec3 startColor = vec3(1.0, 0.0, 0.0);     // Pure red
        vec3 endColor = vec3(0.0, 0.0, 0.8);       // Slightly less intense blue
        vec3 color = mix(startColor, endColor, t);
        
        // Enhanced lighting calculation
        vec3 lightDir = normalize(vec3(1.0, 1.0, 1.0));
        float ambient = 0.3;  // Increased ambient light to show colors better
        float diffuse = max(dot(normalize(vNormal), lightDir), 0.0);
        float lighting = ambient + diffuse * 0.7;   // Adjusted lighting ratio
        
        // Final color with enhanced red preservation
        vec3 finalColor = color * lighting;
        finalColor.r = mix(finalColor.r, color.r, 0.3);  // Preserve some red regardless of lighting
        
        gl_FragColor = vec4(finalColor, 1.0);
    }
    """

# Create shader program
vs_shader = Graphic3d_ShaderObject.CreateFromSource(Graphic3d_TOS_VERTEX, vertex_shader)
fs_shader = Graphic3d_ShaderObject.CreateFromSource(
    Graphic3d_TOS_FRAGMENT, fragment_shader
)
shader_program = Graphic3d_ShaderProgram()
shader_program.AttachShader(vs_shader)
shader_program.AttachShader(fs_shader)

# Apply shader to the shape
aspect = ais_shape.Attributes().ShadingAspect().Aspect()

if shader_program.IsDone():
    aspect.SetShaderProgram(shader_program)
else:
    raise AssertionError("Failed to compile shader program")

# Verify shader program is attached
assert aspect.ShaderProgram() is not None, "Shader program is null"

# Update display
display.Context.Redisplay(ais_shape, True)
display.FitAll()

# Start display
start_display()
```

[image]

closed 2024-11-16 by tpaviot

Bill-XU, 2024-11-16:
@tpaviot 
Thank you very much! So the shader does the work. Have much to learn 🤦‍♂️
Ever asked GPT how, he told me there is no way to achieve it 😓