tpaviot/pythonocc-core #1390: Default Value of linear_deflection

ThisGuyIsNotAJumpingBear, 2024-11-18:
Hi,

I was wondering why the `linear_deflection` parameter in write_stl_file(), line 394 src/Extend/DataExchange.py, is set to 0.9 but claimed to be "default to 0.001" in the docstring? Thanks!

tpaviot, 2024-11-19:
It's a mistake, thank you for the report.

closed 2024-11-19 by tpaviot

ThisGuyIsNotAJumpingBear, 2024-11-19:
Also, may I ask why 0.9 is selected here instead of 0.001? It seems that the current default value is decreasing the precision and making the final shape's quality worse.

tpaviot, 2024-11-19:
I guess that the 0.001 value generates a *very* big file (it was a long time agot the last time I tweaked this part of the code). Default to 0.9 saves disk space and computation time, if you need better accuracy, you can change the default value to a lower one.

Ma-Weijian, 2024-11-19:
Hi there,
I just wonder if the choice of the `linear_deflection` value affect the validity of the final mesh? Just something like setting `linear_deflection` increases the likelihood of obtaining a mesh instead of meeting an exception saying the shape is illegal?

commit 3034d9fe (2024-11-19): Typos in DataExchange, fix #1390