tpaviot/pythonocc-core #1391: Importing issue: OCC.Core.Graphic3d.Graphic3d_Vec3

Bill-XU, 2024-11-19:
Hi,

I am trying to push some variable to shader program.
But while importing Graphic3d_Vec3 (which is the data type of the variable I am using),
the following error occurred.
```powershell
Traceback (most recent call last):
  File "C:\test_shader.py", line 5, in <module>
    from OCC.Core.Graphic3d import (
ImportError: cannot import name 'Graphic3d_Vec3' from 'OCC.Core.Graphic3d' (C:\Program Files\Python311\Lib\site-packages\OCC\Core\Graphic3d.py)
```

I have checked Graphic3d.py, in which there are the following.
```python
# the following typedef cannot be wrapped as is
Graphic3d_Vec3 = NewType("Graphic3d_Vec3", Any)
```
Does this mean that Graphic3d_Vec3 cannot be used in PythonOCC?
If then, how could I push variable to shader program?
Please kindly give your help~~
Thank you in advance!

Best regards,
Bill

Bill-XU, 2024-11-19:
Tried pushing three float variables and assembling them inside the shader program.
But values may be not passed correctly, the color of shader is displayed in black when pushing 1.0, 0.0, and 0.0 for RGB respectively and expecting it be displayed in red.

Bill-XU, 2024-11-19:
> Tried pushing three float variables and assembling them inside the shader program. But values may be not passed correctly, the color of shader is displayed in black when pushing 1.0, 0.0, and 0.0 for RGB respectively and expecting it be displayed in red.

Misunderstood the usage of the GLSL keyword "varying", using "uniform" instead worked well.
So, pushing floats worked well, but still expect that anybody can help advise the way to push a vec3 variable.
Thanks in advance!

Bill

tpaviot, 2024-11-19:
Graphic3d_Vec3 is not currently wrapped.

Bill-XU, 2024-11-19:
> Graphic3d_Vec3 is not currently wrapped.

Thank you. Acknowledged.
I'll continue pushing floats to assemble vec3 inside.