tpaviot/pythonocc-core #1395: Select Topology version on brep export

yann-Mar, 2024-12-06:
Hello,

I am quite unexperienced with python wrappers of c++ code. It seems like the Write version in OCC.Core.BRepTools can point to different version of the c++ write function. 

How would you do to wirte a brep file while specifying the version of the topology because it seems like I can only set the shape and file path in the function ?

Thank you in advance.

Regards,

tpaviot, 2024-12-06:
I'm not sure I fully understand your question/request. What do you mean with "version" of the topology? Is it something that is available in OCCT and is not exposed in the python wrapper? Or is it something else? As far as I know, there's no "version" of the BRep file format used by occt to natively serialize TopoDS_Shapes, did I miss something?

If the problem is to export additional information than topology or geometry (e.g. the "version" or any metadata needed at some point in the workflow), then you have to export the metadata to a separate file, for example a json file, which also stores the brep export (as a string in the json dict, or as a path).

maccardo, 2024-12-13:
When I use the following code I get a message "File was not written with this version of the topology"

import os
import xmltodict
from OCC.Display.SimpleGui import init_display
from OCC.Core.BRepTools import breptools

display, start_display, add_menu, add_function_to_menu = init_display()
script_dir = os.path.dirname(os.path.abspath(__file__))
file_path = os.path.join(script_dir, 'test.xml')
with open(file_path, 'r') as file:
    xml_content = file.read()
xml_dict = xmltodict.parse(xml_content[3:])
shape_data = (xml_dict['root']['model']['shape']).replace('\r','').replace('\n','')
#readResult=breptools.Read(shape,shape_data,builder)
shape=breptools.ReadFromString(shape_data)
display.DisplayShape(shape, update=True)
display.FitAll()
display.Repaint()
start_display()
print('Done..')

maccardo, 2024-12-16:
> I'm not sure I fully understand your question/request. What do you mean with "version" of the topology? Is it something that is available in OCCT and is not exposed in the python wrapper? Or is it [...]

In other words 
brep written by pythonocc-core I see a header of CASCADE Topology V3, (c) Open Cascade
brep written by a foreign third party app, CASCADE Topology V1, (c) Matra-Datavision

I am now curious on compatibility as I see different version and different copyright signatures.
Any thoughts would be appreciated.

Marco

tpaviot, 2024-12-21:
There's an option, in breptools, to choose the topology version when exporting, see https://dev.opencascade.org/doc/refman/html/class_b_rep_tools.html#a4e8058d053fe6924f7c3f357048b517d

Try:
```python

    from OCC.Core.TopTools import TopTools_FormatVersion_VERSION_1
    from OCC.Core.Message import Message_ProgressRange
    breptools.Write(
            my_shape,
            "the_brep_file.brep",
            True,  # export triangles
            True, # export normals
            TopTools_FormatVersion_VERSION_1,  # choss VERSION_2 or VERSION_3
            Message_ProgressRange(), # required to fit the method signature
        )
```