tpaviot/pythonocc-core #1398: breptools.read returns false

amarco, 2024-12-12:
I am using static method breptools.Read and it is always returning false so I never get a shape object.
How do I debug this as the Read provides no telemetry as to why it failed.

tpaviot, 2024-12-14:
The library tells you the .brep file is corrupted. You should not manually tweak this file, it is generated by occt.

closed 2024-12-14 by tpaviot

maccardo, 2024-12-15:
one thing I noticed is the file from the third party is labeled CASCADE Topology V3, but if I make a brep file with the pythonocc-core i get CASCADE Topology V1, at the top of the brep.
Could my problem be python is looking for v1 but I am trying to feed it v3 and hence it fails?

Marco

maccardo, 2024-12-15:
> The library tells you the .brep file is corrupted. You should not manually tweak this file, it is generated by occt.

one thing I noticed is the file from the third party is labeled CASCADE Topology V3, but if I make a brep file with the pythonocc-core i get CASCADE Topology V1, at the top of the brep.
Could my problem be python is looking for v1 but I am trying to feed it v3 and hence it fails?

Marco