tpaviot/pythonocc-core #1399: TopTools_ShapeMapHasher has static methods removed

organic-xml-parser, 2024-12-13:
I noticed that the classes TopTools_ShapeMapHasher and TopTools_OrientedShapeMapHasher have had their static fields removed in this diff: https://github.com/tpaviot/pythonocc-core/commit/40ce519e19fd4bb3cd2dae5e4dc80a55978c2405#diff-5a2c7cbb583b5866446dba51ab9abd3c1ebe92f5712868fc468c417727b7760a

(Relevant file: TopTools.pyi)

I was using these classes so it would be nice if support for those methods would be restored.

organic-xml-parser, 2024-12-14:
Ah I see. Thanks for the explanation. I got it to work for the time being by writing my own swig wrapper to delegate to the operator overloads, since python can't handle function overloading:
```
    static int shape_map_hasher_hash_code(const TopoDS_Shape& shape) {
        TopTools_ShapeMapHasher hasher;
        return hasher(shape);
    }

    static bool shape_map_hasher_equal(const TopoDS_Shape& a, const TopoDS_Shape& b) {
        TopTools_ShapeMapHasher hasher;
        return hasher(a, b);
    }
```
Looks like in future it could work transparently with python's __hash__ and __eq__ methods though!

closed 2024-12-14 by organic-xml-parser

tpaviot, 2024-12-15:
why don't you just use ```hash(shape)```? which is already defined https://github.com/tpaviot/pythonocc-core/blob/master/src/SWIG_files/wrapper/TopoDS.i#L1645

```shell
>>> shp = BRepPrimAPI_MakeTorus(10, 20).Shape()
>>> hash(shp)
886526033189930301
```

organic-xml-parser, 2024-12-18:
Ah, if it works the same way then that is great. I need to do some refactoring since I defined my own "SetPlaceableShape" wrapper class before ;) Thanks!