tpaviot/pythonocc-core #1400: looking for coredataio

maccardo, 2024-12-13:
I am trying to find if coredataio classes are supported in pythonocc-core, I have needs for these equivalnet classes and their data storage io features.
Thanks
Marco

maccardo, 2024-12-13:
sorry for this I am totally turned around on this framework, occ does not have coreiodata classes. Please ignore this question.

closed 2024-12-13 by maccardo

tpaviot, 2024-12-13:
@maccardo excuse this question, but what is coredataio?

maccardo, 2024-12-13:
not a problem. apparently some developers thought to wrap the brep read and writes into a c# class called coredataio and it looked after the the actual file save and reads, with a bunch of base64 encoding, zlib stuff and other data manipulations all of which are causing me a headache. I finally have a string in memory that looks like brep but it has a bunch of \n in it which is now failing the breptools.read method. Do you have any hints no how to clean up my python string that has a bunch of \n in it so it works with breptools.read.

Marco

tpaviot, 2024-12-14:
the .brep file format is not intended to be handled by something else than occt itself. I have no clue to fix a corrupted file.