tpaviot/pythonocc-core #1402: fillet produces strange shapes

Matrixxxxxxxx, 2024-12-25:
[image]
I have filled the top and bottom edges of the shape in the picture, but the result of fillet is a bit strange (not smooth and coherent). I want to know why? Is there a solution? Thank you

Matrixxxxxxxx, 2024-12-25:
The top edge is obtained by simple extrude, and the bottom circular edge is obtained by revolve.

tpaviot, 2024-12-26:
imho this is not related to pythonocc, but rather how the occt library computes fillets. You should report the use case to their issue tracker https://github.com/Open-Cascade-SAS/OCCT/issues. If you do think pythonocc is involved, please submit a python code that can be executed/debugged.

Matrixxxxxxxx, 2024-12-27:
Thanks, I will report this case to OCCT

closed 2024-12-27 by Matrixxxxxxxx