tpaviot/pythonocc-core #1413: Removing the Seam Line in Fully Revolved Shapes

volkenanix, 2025-03-10:
How can I remove or even hide the seam line in the fully revolved shapes?
This seam is also noticeable in a simple sphere.
I want to render the scene in shaded mode and visualize boundary edges but excluding the seam lines.

[image]

Here is a minimal example to reproduce the issue:

```
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeRevol
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeWire, BRepBuilderAPI_MakeEdge
from OCC.Core.gp import gp_Pnt, gp_Ax1, gp_OZ, gp_Circ, gp_Ax2, gp_Dir
from OCC.Core.GC import GC_MakeArcOfCircle
from OCC.Display.SimpleGui import init_display

def create_revolved_solid():
    # Define a half-circle arc as the profile
    radius = 50.0
    circle = gp_Circ(gp_Ax2(gp_Pnt(0, 0, 0), gp_Dir(0, 0, 1)), radius)
    arc = GC_MakeArcOfCircle(circle, gp_Pnt(0, radius, 0), gp_Pnt(0, -radius, 0), True).Value()

    # Create edges from the arc
    arc_edge = BRepBuilderAPI_MakeEdge(arc).Edge()
    
    # Close the profile by adding a vertical line to the Z-axis
    line_edge = BRepBuilderAPI_MakeEdge(gp_Pnt(0, radius, 0), gp_Pnt(0, -radius, 0)).Edge()

    # Create a closed wire
    wire_maker = BRepBuilderAPI_MakeWire()
    wire_maker.Add(arc_edge)
    wire_maker.Add(line_edge)
    wire = wire_maker.Wire()

    # Define the axis of revolution (Z-axis)
    revolution_axis = gp_Ax1(gp_Pnt(0, 2*radius, 0), gp_Dir(1, 0, 0))

    # Perform the revolution
    revolved_solid = BRepPrimAPI_MakeRevol(wire, revolution_axis).Shape()
    
    return revolved_solid

if __name__ == "__main__":
    shape = create_revolved_solid()

    display, start_display, _, _ = init_display()
    display.DisplayShape(shape, update=True)
    display.FitAll()
    start_display()
```

volkenanix, 2025-03-15:
@jf--- 
Is there any way to hide the seam? I checked in commercial software like SolidWorks, the seam line exists but it is not shown by default.
I looked for the solution in Open CASCADE reference and found a property called `SetUnFreeBoundaryDraw(false)` which enables or disables drawing of shared boundaries for shading presentations. I tried it but the seam was still shown.

gkv311, 2025-03-19:
The displaying of seem edges could be managed via `Prs3d_Drawer::FaceBoundaryUpperContinuity()` property.

```.py
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeRevol
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeWire, BRepBuilderAPI_MakeEdge
from OCC.Core.gp import gp_Pnt, gp_Ax1, gp_OZ, gp_Circ, gp_Ax2, gp_Dir
from OCC.Core.GC import GC_MakeArcOfCircle
from OCC.Core.GeomAbs import *
from OCC.Display.SimpleGui import init_display

def create_revolved_solid():
    # Define a half-circle arc as the profile
    radius = 50.0
    circle = gp_Circ(gp_Ax2(gp_Pnt(0, 0, 0), gp_Dir(0, 0, 1)), radius)
    arc = GC_MakeArcOfCircle(circle, gp_Pnt(0, radius, 0), gp_Pnt(0, -radius, 0), True).Value()

    # Create edges from the arc
    arc_edge = BRepBuilderAPI_MakeEdge(arc).Edge()
    
    # Close the profile by adding a vertical line to the Z-axis
    line_edge = BRepBuilderAPI_MakeEdge(gp_Pnt(0, radius, 0), gp_Pnt(0, -radius, 0)).Edge()

    # Create a closed wire
    wire_maker = BRepBuilderAPI_MakeWire()
    wire_maker.Add(arc_edge)
    wire_maker.Add(line_edge)
    wire = wire_maker.Wire()

    # Define the axis of revolution (Z-axis)
    revolution_axis = gp_Ax1(gp_Pnt(0, 2*radius, 0), gp_Dir(1, 0, 0))

    # Perform the revolution
    revolved_solid = BRepPrimAPI_MakeRevol(wire, revolution_axis).Shape()
    
    return revolved_solid

if __name__ == "__main__":
    shape = create_revolved_solid()

    display, start_display, _, _ = init_display()

    # adjust default display properties (Prs3d_Drawer in AIS_InteractiveContext)
    display.Context.DefaultDrawer().SetFaceBoundaryDraw(True)
    display.Context.DefaultDrawer().SetFaceBoundaryUpperContinuity(GeomAbs_G2)

    display.DisplayShape(shape, update=True)
    display.FitAll()
    start_display()
```