tpaviot/pythonocc-core #1417: Build failure when PYTHONOCC_MESHDS_NUMPY=OFF

Lgt2x, 2025-03-19:
In version 7.8.1.1, building `pythonocc-core` fails when the `PYTHONOCC_MESHDS_NUMPY` CMake option is disabled:

````
pythonocc-core/build/Unix/x86_64--/TShortPYTHON_wrap.cxx:4429:10: fatal error: numpy/arrayobject.h: No such file or directory
 4429 | #include <numpy/arrayobject.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

This fails because numpy headers are included even when the option is off, but the `find_package(Python3)` [does not look for](https://github.com/tpaviot/pythonocc-core/blob/master/CMakeLists.txt#L104) the `NumPy` component in this case.

I assume this is related to #1381 adding Numpy support to more modules. This problem was not caught by CI builds, because all of them build with Numpy enabled.

fyi @mwestphal

sbianchinitweddle, 2025-09-01:
In my Dockerfile i used

```
RUN apt-get install -y python3-numpy
ENV CPLUS_INCLUDE_PATH=/usr/lib/python3/dist-packages/numpy/core/include
```

to solve this issue.

PhilHay, 2025-10-02:
The problem still exists in 7.9.0
MESHDS_NUMPY is never disabled, it is (strangely) included in the option PYTHONOCC_WRAP_VISU.

Extract from Cmakefile.txt
```
346   ##########################
347   # MeshDS module addition #
348   ##########################
349   execute_process(COMMAND ${CMAKE_COMMAND} -E make_directory src/MeshDataSource)
350   set(FILE ${SWIG_FILES_PATH}/MeshDS.i)
351   set_source_files_properties(${FILE} PROPERTIES CPLUSPLUS ON)
352   include_directories(${CMAKE_CURRENT_SOURCE_DIR}/src/MeshDataSource)
353   set(MESHDATASOURCE_SOURCE_FILES
354   ${FILE}
355   ${CMAKE_CURRENT_SOURCE_DIR}/src/MeshDataSource/MeshDataSource.cpp)
356 
357   swig_add_library(MeshDS LANGUAGE python SOURCES ${MESHDATASOURCE_SOURCE_FILES} TYPE MODULE)
358   swig_link_libraries(MeshDS ${OCCT_MODEL_LIBRARIES} ${OCCT_VISUALIZATION_LIBRARIES} Python3::Module)
359 
360 endif(PYTHONOCC_WRAP_VISU)
```

tpaviot, 2026-09-22:
Fixed

closed 2026-09-22 by tpaviot