tpaviot/pythonocc-core #1435: ShapeTool.GetSubShapes() returns empty list in new version

aruig666, 2025-05-23:
After updating pythonocc-core to 7.9.0, I found that using ShapeTool to extract subshapes (e.g. faces, edges, etc.) from a loaded model returns an empty list, whereas it worked correctly in previous versions.

`
import pathlib
from OCC.Core.XCAFApp import XCAFApp_Application
from OCC.Core.TDocStd import TDocStd_Document
from OCC.Core.STEPCAFControl import STEPCAFControl_Reader
from OCC.Core.XCAFApp import XCAFApp_Application
from OCC.Core.TDocStd import TDocStd_Document
from OCC.Core.XCAFDoc import XCAFDoc_DocumentTool, XCAFDoc_DocumentTool
from OCC.Core.Quantity import Quantity_Color, Quantity_TOC_RGB
from OCC.Core.TDF import TDF_LabelSequence
from OCC.Core.TDF import TDF_Label
from OCC.Core.TopExp import TopExp_Explorer
from OCC.Core.TopAbs import TopAbs_FACE
from OCC.Core.BRep import BRep_Builder
from OCC.Core.TopoDS import TopoDS_Compound, TopoDS_Shape
from OCC.Extend.DataExchange import read_step_file_with_names_colors
import random

app = XCAFApp_Application.GetApplication()
doc = TDocStd_Document("pythonocc-doc")
app.NewDocument("pythonocc-doc", doc)

reader = STEPCAFControl_Reader()
reader.SetColorMode(True)
reader.SetNameMode(True)
reader.SetLayerMode(True)
stepname = pathlib.Path(r"output.step")

filename = str(stepname)
status = reader.ReadFile(filename)

if status:
    reader.Transfer(doc)
    print("STEP file loaded successfully.")
else:
    print("Failed to load STEP file.")

shape_tool = XCAFDoc_DocumentTool.ShapeTool(doc.Main())
color_tool = XCAFDoc_DocumentTool.ColorTool(doc.Main())

root_labels = TDF_LabelSequence()
shape_tool.GetFreeShapes(root_labels)
print(root_labels.Length())
if root_labels.Length() > 0:
    root_label = root_labels.Value(1)
    sub_labels = TDF_LabelSequence()
    shape_tool.GetSubShapes(root_label, sub_labels)
    print(sub_labels.Length())
`

return 
`
STEP file loaded successfully.
1
0
`

pythonocc-core version: 7.7.9
Python version: 3.10

[output.txt](https://github.com/user-attachments/files/20406897/output.txt)

Jair-02, 2025-06-12:
I use my step2glb service with pythonocc-core 7.9.0, transfer your output.step file. 
The result like this.

[image]

tpaviot, 2026-09-22:
ShapeTool.GetSubShapes() does not return all the faces or edges of a shape: it returns the sub-shapes that have their own label in the XCAF document. The STEP reader creates such labels only for sub-shapes with attributes (colors, names, layers).

The output.step you attached has no colors and no sub-shape names (it has no STYLED_ITEM entities), so an empty list is the expected result. I checked it with OCCT 8.0.1: a STEP file with colored faces returns them through GetSubShapes.

To get all the faces of a shape, explore the shape itself:

```python
from OCC.Extend.TopologyUtils import TopologyExplorer
shape = shape_tool.GetShape(root_labe
faces = list(TopologyExplorer(shape).faces())
```

If your file had colored faces before you wrote it with 7.9, the colors may have been lost when writing it. Feel free to reopen with the original

closed 2026-09-22 by tpaviot