tpaviot/pythonocc-core #1444: Location not updated by `BRep_Tool.Triangulation`?

inducer, 2025-07-11:
I am reimplementing the logic in the [upstream STL writer](https://github.com/Open-Cascade-SAS/OCCT/blob/508700117cf4d41b99087deed2b05f93e751e5cf/src/DataExchange/TKDESTL/StlAPI/StlAPI_Writer.cxx#L69-L116) in pythonocc-core in order to obtain a surface mesh without having to resort to File I/O, and it seems that calling `BRep_Tool.Triangulation` should update the `TopLoc_Location` it is passed, but, at least according to its `DumpJSON` method, this is not happening. Here's the core of my traversal loop:
```py
    exp = TopExp_Explorer(shape, TopAbs_FACE)
    while exp.More():
        face = exp.Current()
        if face.Orientation() == TopAbs_REVERSED:
            print("FIXME Reversed face")

        loc = TopLoc_Location()
        tri = BRep_Tool.Triangulation(face, loc)
        if tri:
            trsf = loc.Transformation()
            for i in range(tri.NbNodes()):
                p = tri.Node(1+i)
                p.Transform(trsf)
                nodes.append([p.X(), p.Y(), p.Z()])
```
[A full self-contained example](https://gist.github.com/inducer/591ee58ccf03bcb8c262cef513dee85b) is also available.

In the full example, I am trying to mesh a box. STL export (ostensibly using the OCCT C++ code above) works fine, but with my reimplementation, I get a mesh that looks like this:

[image]

I.e. all faces of the box seem to coincide in one plane, seemingly consistent with lack of transformation information.

tpaviot, 2025-09-19:
Not sure the problem comes from the point coordinates, maybe the face definition is wrong.

tpaviot, 2026-09-22:
The problem is the triangle indices: tri.Triangle(j).Get() returns indices local to the face (1 to tri.NbNodes()), but you append the nodes of all the faces to a single list. Every triangle then refers to the nodes of the first face. Offset them by the number of nodes already collected, as StlAPI_Writer does, and swap two indices of reversed faces to keep the normals outward:

```python
first = len(nodes)
for i in range(1, tri.NbNodes() + 1):
    p = tri.Node(i).Transformed(loc.Transformation())
    nodes.append([p.X(), p.Y(), p.Z()])
for j in range(1, tri.NbTriangles() + 1):
    n1, n2, n3 = tri.Triangle(j).Get()
     if face.Orientation() == TopAbs_REVERSED:
        n2, n3 = n3, n2
    vertices.append([first + n1 - 1, first + n2 - 1, first + n3 - 1])
```

closed 2026-09-22 by tpaviot